THU-DSP-LAB/llvm-project #1: fails to compile vector_dot_product testcase

yangkex, 2023-02-28:
my environment is ubuntu 20.04, and I compile example `vecadd.cl` successfully, but for this testcase it fails. 
command is:
` /opt/llvm-project/install/bin/clang  -cl-std=CL2.0 -target riscv32 -mcpu=ventus-gpgpu -O1 -S example1.cl -o example1.s`
source file `example1.cl` is:
```
#define USE_VECTOR_DATATYPES

__kernel void dot_product (__global const float4 *a,
             __global const float4 *b, __global float *c)
{
  int gid = get_global_id(0);

#ifndef USE_VECTOR_DATATYPES
  /* This version is to smoke test the autovectorization.
     Tries to create parallel regions with nice memory
     access pattern etc. so it gets autovectorizer. */
  /* This parallel region does not vectorize with the
     loop vectorizer because it accesses vector datatypes.
     Perhaps with SLP/BB vectorizer.*/

  float ax = a[gid].x;
  float ay = a[gid].y;
  float az = a[gid].z;
  float aw = a[gid].w;

  float bx = b[gid].x,
      by = b[gid].y,
      bz = b[gid].z,
      bw = b[gid].w;

  barrier(CLK_LOCAL_MEM_FENCE);

  /* This parallel region should vectorize. */
  c[gid] = ax * bx;
  c[gid] += ay * by;
  c[gid] += az * bz;
  c[gid] += aw * bw;

#else
  float4 prod = a[gid] * b[gid];
  c[gid] = prod.x + prod.y + prod.z + prod.w;
#endif

}
```
the report is:
```
clang: /opt/llvm-project/llvm/lib/Target/RISCV/RISCVISelDAGToDAG.cpp:879: bool llvm::RISCVDAGToDAGISel::SelectAddrRegReg(llvm::SDValue, llvm::SDValue&, llvm::SDValue&): Assertion `0 && "TODO"' failed.
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace, preprocessed source, and associated run script.
Stack dump:
0.      Program arguments: /opt/llvm-project/install/bin/clang -cl-std=CL2.0 -target riscv32 -mcpu=ventus-gpgpu -O1 -S example1.cl -o example1.s
1.      <eof> parser at end of file
2.      Code generation
3.      Running pass 'Function Pass Manager' on module 'example1.cl'.
4.      Running pass 'RISCV DAG->DAG Pattern Instruction Selection' on function '@dot_product'
 #0 0x00007f81ee6e3298 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) /opt/llvm-project/llvm/lib/Support/Unix/Signals.inc:567:22
 #1 0x00007f81ee6e3353 PrintStackTraceSignalHandler(void*) /opt/llvm-project/llvm/lib/Support/Unix/Signals.inc:641:1
 #2 0x00007f81ee6e0f5e llvm::sys::RunSignalHandlers() /opt/llvm-project/llvm/lib/Support/Signals.cpp:104:20
 #3 0x00007f81ee6e2ab9 llvm::sys::CleanupOnSignal(unsigned long) /opt/llvm-project/llvm/lib/Support/Unix/Signals.inc:366:31
 #4 0x00007f81ee52a87b (anonymous namespace)::CrashRecoveryContextImpl::HandleCrash(int, unsigned long) /opt/llvm-project/llvm/lib/Support/CrashRecoveryContext.cpp:77:5
 #5 0x00007f81ee52adf6 CrashRecoverySignalHandler(int) /opt/llvm-project/llvm/lib/Support/CrashRecoveryContext.cpp:398:1
 #6 0x00007f81edea3090 (/lib/x86_64-linux-gnu/libc.so.6+0x43090)
 #7 0x00007f81edea300b raise (/lib/x86_64-linux-gnu/libc.so.6+0x4300b)
 #8 0x00007f81ede82859 abort (/lib/x86_64-linux-gnu/libc.so.6+0x22859)
 #9 0x00007f81ede82729 (/lib/x86_64-linux-gnu/libc.so.6+0x22729)
#10 0x00007f81ede93fd6 (/lib/x86_64-linux-gnu/libc.so.6+0x33fd6)
#11 0x00007f81f9313dfa llvm::RISCVDAGToDAGISel::SelectAddrRegReg(llvm::SDValue, llvm::SDValue&, llvm::SDValue&) /opt/llvm-project/llvm/lib/Target/RISCV/RISCVISelDAGToDAG.cpp:886:21
#12 0x00007f81f931e2c3 llvm::RISCVDAGToDAGISel::CheckComplexPattern(llvm::SDNode*, llvm::SDNode*, llvm::SDValue, unsigned int, llvm::SmallVectorImpl<std::pair<llvm::SDValue, llvm::SDNode*>>&) /opt/llvm-project/build/lib/Target/RISCV/RISCVGenDAGISel.inc:17100:78
#13 0x00007f81eca68299 llvm::SelectionDAGISel::SelectCodeCommon(llvm::SDNode*, unsigned char const*, unsigned int) /opt/llvm-project/llvm/lib/CodeGen/SelectionDAG/SelectionDAGISel.cpp:3119:31
#14 0x00007f81f931974e llvm::RISCVDAGToDAGISel::SelectCode(llvm::SDNode*) /opt/llvm-project/build/lib/Target/RISCV/RISCVGenDAGISel.inc:16112:1
[... 28 lines cut ...]
#43 0x00007f81ee52b008 llvm::CrashRecoveryContext::RunSafely(llvm::function_ref<void ()>) /opt/llvm-project/llvm/lib/Support/CrashRecoveryContext.cpp:434:10
#44 0x00007f81f3bb6187 clang::driver::CC1Command::Execute(llvm::ArrayRef<std::optional<llvm::StringRef>>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>>*, bool*) const /opt/llvm-project/clang/lib/Driver/Job.cpp:428:7
#45 0x00007f81f3b39bf8 clang::driver::Compilation::ExecuteCommand(clang::driver::Command const&, clang::driver::Command const*&, bool) const /opt/llvm-project/clang/lib/Driver/Compilation.cpp:200:22
#46 0x00007f81f3b39f86 clang::driver::Compilation::ExecuteJobs(clang::driver::JobList const&, llvm::SmallVectorImpl<std::pair<int, clang::driver::Command const*>>&, bool) const /opt/llvm-project/clang/lib/Driver/Compilation.cpp:254:62
#47 0x00007f81f3b52acb clang::driver::Driver::ExecuteCompilation(clang::driver::Compilation&, llvm::SmallVectorImpl<std::pair<int, clang::driver::Command const*>>&) /opt/llvm-project/clang/lib/Driver/Driver.cpp:1816:28
#48 0x0000561f277f4196 clang_main(int, char**) /opt/llvm-project/clang/tools/driver/driver.cpp:520:39
#49 0x0000561f27826c5e main /opt/llvm-project/build/tools/clang/tools/driver/clang-driver.cpp:11:63
#50 0x00007f81ede84083 __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x24083)
#51 0x0000561f277f125e _start (/opt/llvm-project/install/bin/clang+0x4f25e)
clang-16: error: clang frontend command failed with exit code 134 (use -v to see invocation)
clang version 16.0.0 (https://github.com/THU-DSP-LAB/llvm-project.git 60a504f481826a3e1fa84aa120786110e8714944)
Target: riscv32
Thread model: posix
InstalledDir: /opt/llvm-project/install/bin
clang-16: note: diagnostic msg:
********************

PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
Preprocessed source(s) and associated run script(s) are located at:
clang-16: note: diagnostic msg: /tmp/example1-4fb1fd.cl
clang-16: note: diagnostic msg: /tmp/example1-4fb1fd.sh
clang-16: note: diagnostic msg:

********************
```

zhoujingya, 2023-02-28:
@yangkex  will later, check

zhoujingya, 2023-03-01:
@yangkex solved by this commit https://github.com/THU-DSP-LAB/llvm-project/commit/ff4e666b80102b03b95e0fbf514583e4c0a70d6b, you can check it,

closed 2023-03-01 by yangkex