THU-DSP-LAB/llvm-project #16: [CTS] basic/if failed

guaguabao123, 2023-09-11:
1. run test_basic if  failed 
2. simple change  case to test
    A.  set 32 threads , run test_basic if 32
    B.   
[image]
3. run failed
    input data  has 0x1 ,but output data is 0x7fffffff
    
[image]

zhoujingya, 2023-09-12:
做了一些测试，看了一下spike的输出，发现有指令报访存指令的错误,看这个[链接](https://paste.ubuntu.com/p/VB5wmQTz7y/)

zhoujingya, 2023-09-12:
发现core0的结果是对的，看这个[链接]( https://paste.ubuntu.com/p/DKFStgXkFR/)

guaguabao123, 2023-09-13:
dump文件[https://paste.ubuntu.com/p/chZ87V4HPz/](url)
log文件[https://paste.ubuntu.com/p/Zw77bNjngJ/](url)

zhoujingfighting, 2023-09-13:
目前总结得出，是软件这边的bug，得优化访存行为

zhoujingya, 2023-09-13:
[image]
确定是join指令这边出问题了

zhoujingya, 2023-09-15:
__kernel void test_if(__global int *src, __global int *dst)
{
    int  tid = get_global_id(0);

    if (src[tid] == 0)
        dst[tid] = 0x12345678;
    else if (src[tid] == 1)
        dst[tid] = 0x23456781;
    else if (src[tid] == 2)
        dst[tid] = 0x34567812;
    else if (src[tid] == 3)
        dst[tid] = 0x45678123;
    else if (src[tid] == 4)
        dst[tid] = 0x56781234;
    else if (src[tid] == 5)
        dst[tid] = 0x67812345;
    else if (src[tid] == 6)
        dst[tid] = 0x78123456;
    else if (src[tid] == 7)
        dst[tid] = 0x81234567;
    else
        dst[tid] = 0x7FFFFFFF;

}

zhoujingya, 2023-09-25:
之前的提交有点问题:https://github.com/THU-DSP-LAB/llvm-project/commit/4bfde6aa71c9f44778990bec81a3e18b6808a453

zhoujingya, 2023-09-25:
solve by this commit 58e57e868b00411cb3717cc9f1479cd5bad670a9

closed 2023-10-25 by zhoujingya