THU-DSP-LAB/llvm-project #20: Discussion about test

zhoujingya, 2023-09-14:
目前的测试其实包括了很多模块，但是主要还是针对pocl和spike的功能测试，当前的测试比较分散，没有加入CTS，只有pocl和rodinia的部分测试，个人觉得不是很完善，应该把CTS通过的部分也加入进去，以下想法:

1 : 创建一个子仓库，里面存所有待测试的可执行文件，文件层级按目前的仓库来划分。类似
pocl
	vecadd
	matadd
gpurodia
	nn
	gaussian
CTS
	testconformance
		basic
				if
				kernel_call_kernel
		compiler
		...

---
每次有通过的新用例可以往这个子仓库push,然后写一个脚本来跑测试，以后每次PR之前，都做一个这个测试，保证更改之后之前的测试都能过，这是一个大致的指导思路

Jules-Kong, 2023-09-15:
关于回归测试：
1.  现有用例，还存在简陋的构建和验证问题；
2.  关于CTS，可以延续之前通过用例加入的规则，但要考虑时间成本；（门禁方式，定时触发，手动触发）

zhoujingya, 2023-09-21:
https://paste.ubuntu.com/p/TytRsbyJBB/
https://paste.ubuntu.com/p/yT8YNKF7VG/
参考这两个脚本
```
TEST
├── cts
│   ├── test_api
│   ├── test_basic
│   └── test_compiler
├── pocl
│   ├── matadd
│   └── vecadd
└── rodiania
```

zhoujingya, 2023-09-22:
build.sh
```SHELL
#!/usr/bin/bash
DIR=$(cd "$(dirname "${0}")" &> /dev/null && (pwd -W 2> /dev/null || pwd))
source ${DIR}/data.sh
# 指定要搜索的目录路径
pocl_directory="/home/triton/Codes/OpenCL-CTS/TEST/pocl"
cts_directory="/home/triton/Codes/OpenCL-CTS/TEST/cts"
# 使用 find 命令列出目录层级下的所有文件
pocl_executables=($(find "$pocl_directory" -type f -executable -print0 | xargs -0))
cts_executables=($(find "$cts_directory" -type f -executable -print0 | xargs -0))
#
export LD_LIBRARY_PATH=/work/ventus-llvm/install/lib
export OCL_ICD_VENDORS=/work/ventus-llvm/install/lib/libpocl.so
export POCL_DEVICES="ventus"

# 遍历数组并执行每个可执行文件
for file in "${pocl_executables[@]}"; do
  echo "----"
  if [ -x "$file" ]; then
    echo "Running $file"
    "$file" >& /dev/null
    if [ $? -eq 0 ]; then
      echo "$file execution successful"
    else
      echo "Execution failed with exit status $exit_status"
      exit $?
    fi
  else
    echo "$file is not executable or does not exist"
  fi
done

run_cts_basic() {

  for element in "${cts_basic_test[@]}"
    do
      echo "Running $1 $element"
      $1 $element 32 >& /dev/null
      if [ $? -eq 0 ]; then
        echo "$1 $element execution successful"
      else
        echo "Execution failed with $1 $element"
        exit $?
      fi
    done
}

run_cts_compiler() {

  for element in "${cts_compiler_test[@]}"
    do
      echo "Running $1 $element"
      $1 $element 32 >& /dev/null
      if [ $? -eq 0 ]; then
        echo "$1 $element execution successful"
      else
        echo "Execution failed with $1 $element"
        exit $?
      fi
    done
}

run_cts_api() {

  for element in "${cts_api_test[@]}"
    do
      echo "Running $1 $element"
      $1 $element 32 >& /dev/null
      if [ $? -eq 0 ]; then
        echo "$1 $element execution successful"
      else
        echo "Execution failed with $1 $element"
      fi
    done
}

# CTS的测试不太一样，需要特别处理
for file in "${cts_executables[@]}"; do
  echo "----"
  if [[ "$file" == */test_basic ]]; then
    run_cts_basic $file
  elif [[ "$file" == */test_compiler ]]; then
    run_cts_compiler $file
  elif [[ "$file" == */test_api ]]; then
    run_cts_api $file
  else
    echo "$file is not executable or does not exist"
  fi
done
```

zhoujingya, 2023-09-22:
data.sh

```shell
cts_basic_test=(
  if
  loop
  constant
  constant_source
  barrier
  wg_barrier
  arrayreadwrite
  # bufferreadwriterect
  # arraycopy
  createkernelsinprogram
  explicit_s2v_double
  host_numeric_constants
  get_global_offset
  global_linear_id
  pointer_cast
  # kernel_numeric_constants
  kernel_call_kernel_function
  kernel_memory_alignment_global
  kernel_memory_alignment_constant
  # kernel_memory_alignment_local
  # fpmath
  # sizeof
  # vload_global
  # vload_local
  # vload_constant
  # vload_private
  # explicit_s2v_char
  # explicit_s2v_uchar
  # explicit_s2v_short
  # explicit_s2v_ushort
  # explicit_s2v_int
  # explicit_s2v_uint
  # explicit_s2v_long
  # explicit_s2v_ulong
  # explicit_s2v_float
  # explicit_s2v_double
  # vec_type_hint
  # vector_creation
  # vector_hint
)

cts_compiler_test=(
  load_program_source
  load_multistring_source
  load_two_kernel_source
  load_null_terminated_source
  load_null_terminated_multi_line_source
  load_null_terminated_partial_multi_line_source
  load_discreet_length_source
  get_program_source
  get_program_build_info
  get_program_info
  large_compile
  async_build
  options_build_optimizations
  preprocessor_define_udef
  preprocessor_include
  preprocessor_line_error
  preprocessor_pragma
  image_macro
)

cts_api_test=(
  get_platform_info
  get_context_info
  get_device_info
  binary_get
  kernel_required_group_size
  release_kernel_order
  release_during_execute
  load_single_kernel
  load_two_kernels
  load_two_kernels_in_one
  load_two_kernels_manually
  get_program_info_kernel_names
  # get_kernel_arg_info : too long
  create_kernels_in_program
  get_kernel_info
  kernel_private_memory_size
  execute_kernel_local_sizes
  set_kernel_arg_by_index
  set_kernel_arg_constant
  set_kernel_arg_struct_array
  kernel_global_constant
  # kernel_attributes : too long
  min_max_thread_dimensions
  min_max_work_items_sizes
  min_max_work_group_size
  min_max_mem_alloc_size
  min_max_compute_units
  min_max_address_bits
  min_max_single_fp_config
  min_max_double_fp_config
  min_max_kernel_preferred_work_group_size_multiple
  min_max_language_version
  create_context_from_type
  platform_extensions
  get_platform_ids
  bool_type
  retain_queue_single
  retain_queue_multiple
  retain_mem_object_single
  retain_mem_object_multiple
  retain_mem_object_set_kernel_arg
  mem_object_destructor_callback
  null_buffer_arg
  negative_get_platform_info
  negative_get_platform_ids
)
```

Mifuns, 2023-10-26:
更新后的data.sh
`cts_basic_test=(
  hostptr           
  # fpmath              
  intmath_int         
  intmath_int2        
  intmath_int4        
  # intmath_long        
  # intmath_long2       
  # intmath_long4           
  # hiloeo              
  if                  
  # sizeof              
  loop                
  pointer_cast        
  # local_arg_def       
  # local_kernel_def    
  # local_kernel_scope  
  constant            
  constant_source     
  readimage           
  readimage_int16     
  readimage_fp32      
  writeimage          
  writeimage_int16    
  writeimage_fp32     
  mri_one             
  mri_multiple        
  image_r8            
  barrier             
  wg_barrier          
  # int2fp              
  # fp2int              
  imagereadwrite      
  imagereadwrite3d    
  readimage3d         
  readimage3d_int16   
  readimage3d_fp32    
  # bufferreadwriterect 
  arrayreadwrite      
  # arraycopy           
  imagearraycopy      
  imagearraycopy3d    
  imagecopy           
  imagecopy3d         
  imagerandomcopy     
  arrayimagecopy      
  arrayimagecopy3d    
  imagenpot           
  # vload_global        
  # vload_local         
  # vload_constant      
  # vload_private       
  # vstore_global       
  # vstore_local        
  # vstore_private      
  createkernelsinprogram  
  imagedim_pow2      
  imagedim_non_pow2   
  image_param        
  image_multipass_integer_coord 
  image_multipass_float_coord   
  # explicit_s2v_char   
  # explicit_s2v_uchar  
  # explicit_s2v_short  
  # explicit_s2v_ushort 
  # explicit_s2v_int    
  # explicit_s2v_uint   
  # explicit_s2v_long   
  # explicit_s2v_ulong  
  # explicit_s2v_float  
  explicit_s2v_double 
  # enqueue_map_buffer  
  enqueue_map_image   
  # work_item_functions 
  # astype              
  # async_copy_global_to_local          
  # async_copy_local_to_global          
  # async_strided_copy_global_to_local  
  # async_strided_copy_local_to_global  
  async_copy_global_to_local2D        
  async_copy_local_to_global2D        
  async_copy_global_to_local3D        
  async_copy_local_to_global3D        
  async_work_group_copy_fence_import_after_export_aliased_local               
  async_work_group_copy_fence_import_after_export_aliased_global              
  async_work_group_copy_fence_import_after_export_aliased_global_and_local    
  async_work_group_copy_fence_export_after_import_aliased_local               
  async_work_group_copy_fence_export_after_import_aliased_global              
  async_work_group_copy_fence_export_after_import_aliased_global_and_local    
  # prefetch                          
  kernel_call_kernel_function       
  host_numeric_constants            
  # kernel_numeric_constants          
  # kernel_limit_constants            
  # kernel_preprocessor_macros        
  # parameter_types                   
  # vector_creation                   
  # vector_swizzle                    
  # vec_type_hint                     
  # kernel_memory_alignment_local     
  kernel_memory_alignment_global    
  kernel_memory_alignment_constant  
  # kernel_memory_alignment_private   
  # progvar_prog_scope_misc           
  # progvar_prog_scope_uninit         
  # progvar_prog_scope_init           
  # progvar_func_scope                
  # global_work_offsets               
  get_global_offset                 
  global_linear_id                  
  # local_linear_id                   
  # enqueued_local_size               
  simple_read_image_pitch           
  simple_write_image_pitch          
  # get_linear_ids                    
  rw_image_access_qualifier         
)

cts_compiler_test=(
  load_program_source
  load_multistring_source
  load_two_kernel_source
  load_null_terminated_source
  load_null_terminated_multi_line_source
  load_null_terminated_partial_multi_line_source
  load_discreet_length_source
  get_program_source
  get_program_build_info
  get_program_info
  large_compile
  async_build
  options_build_optimizations
  preprocessor_define_udef
  preprocessor_include
  preprocessor_line_error
  preprocessor_pragma
  image_macro
)

cts_api_test=(
  get_platform_info
  get_context_info
  get_device_info
  binary_get
  kernel_required_group_size
  release_kernel_order
  release_during_execute
  load_single_kernel
  load_two_kernels
  load_two_kernels_in_one
  load_two_kernels_manually
  get_program_info_kernel_names
  # get_kernel_arg_info : too long
  create_kernels_in_program
  get_kernel_info
  kernel_private_memory_size
  execute_kernel_local_sizes
  set_kernel_arg_by_index
  set_kernel_arg_constant
  set_kernel_arg_struct_array
  kernel_global_constant
  # kernel_attributes : too long
  min_max_thread_dimensions
  min_max_work_items_sizes
  min_max_work_group_size
  min_max_mem_alloc_size
  min_max_compute_units
  min_max_address_bits
  min_max_single_fp_config
  min_max_double_fp_config
  min_max_kernel_preferred_work_group_size_multiple
  min_max_language_version
  create_context_from_type
  platform_extensions
  get_platform_ids
  bool_type
  retain_queue_single
  retain_queue_multiple
  retain_mem_object_single
  retain_mem_object_multiple
  retain_mem_object_set_kernel_arg
  mem_object_destructor_callback
  null_buffer_arg
  negative_get_platform_info
  negative_get_platform_ids
)
`

Mifuns, 2023-10-26:
更新后的data.sh

cts_basic_test=(
  hostptr           
  \# fpmath              
  intmath_int         
  intmath_int2        
  intmath_int4        
  \# intmath_long        
  \# intmath_long2       
  \# intmath_long4           
  \# hiloeo              
  if                  
  \# sizeof              
  loop                
  pointer_cast        
  \# local_arg_def       
  \# local_kernel_def    
  \# local_kernel_scope  
  constant            
  constant_source     
  readimage           
  readimage_int16     
  readimage_fp32      
  writeimage          
  writeimage_int16    
  writeimage_fp32     
  mri_one             
  mri_multiple        
  image_r8            
  barrier             
  wg_barrier          
  \# int2fp              
  \# fp2int              
  imagereadwrite      
  imagereadwrite3d    
  readimage3d         
  readimage3d_int16   
  readimage3d_fp32    
  \# bufferreadwriterect 
  arrayreadwrite      
  \# arraycopy           
  imagearraycopy      
  imagearraycopy3d    
  imagecopy           
  imagecopy3d         
  imagerandomcopy     
  arrayimagecopy      
  arrayimagecopy3d    
  imagenpot           
  \# vload_global        
  \# vload_local         
  \# vload_constant      
  \# vload_private       
  \# vstore_global       
  \# vstore_local        
  \# vstore_private      
  createkernelsinprogram  
  imagedim_pow2      
  imagedim_non_pow2   
  image_param        
  image_multipass_integer_coord 
  image_multipass_float_coord   
  \# explicit_s2v_char   
  \# explicit_s2v_uchar  
  \# explicit_s2v_short  
  \# explicit_s2v_ushort 
  \# explicit_s2v_int    
  \# explicit_s2v_uint   
  \# explicit_s2v_long   
  \# explicit_s2v_ulong  
  \# explicit_s2v_float  
  explicit_s2v_double 
  \# enqueue_map_buffer  
  enqueue_map_image   
  \# work_item_functions 
  \# astype              
  \# async_copy_global_to_local          
  \# async_copy_local_to_global          
  \# async_strided_copy_global_to_local  
  \# async_strided_copy_local_to_global  
  async_copy_global_to_local2D        
  async_copy_local_to_global2D        
  async_copy_global_to_local3D        
  async_copy_local_to_global3D        
  async_work_group_copy_fence_import_after_export_aliased_local               
  async_work_group_copy_fence_import_after_export_aliased_global              
  async_work_group_copy_fence_import_after_export_aliased_global_and_local    
  async_work_group_copy_fence_export_after_import_aliased_local               
  async_work_group_copy_fence_export_after_import_aliased_global              
  async_work_group_copy_fence_export_after_import_aliased_global_and_local    
  \# prefetch                          
  kernel_call_kernel_function       
  host_numeric_constants            
  \# kernel_numeric_constants          
  \# kernel_limit_constants            
  \# kernel_preprocessor_macros        
  \# parameter_types                   
  \# vector_creation                   
  \# vector_swizzle                    
  \# vec_type_hint                     
  \# kernel_memory_alignment_local     
  kernel_memory_alignment_global    
  kernel_memory_alignment_constant  
  \# kernel_memory_alignment_private   
  \# progvar_prog_scope_misc           
  \# progvar_prog_scope_uninit         
  \# progvar_prog_scope_init           
  \# progvar_func_scope                
  \# global_work_offsets               
  get_global_offset                 
  global_linear_id                  
  \# local_linear_id                   
  \# enqueued_local_size               
  simple_read_image_pitch           
  simple_write_image_pitch          
  \# get_linear_ids                    
  rw_image_access_qualifier         
)

cts_compiler_test=(
  load_program_source
  load_multistring_source
  load_two_kernel_source
  load_null_terminated_source
  load_null_terminated_multi_line_source
  load_null_terminated_partial_multi_line_source
  load_discreet_length_source
  get_program_source
  get_program_build_info
  get_program_info
  large_compile
  async_build
  options_build_optimizations
  preprocessor_define_udef
  preprocessor_include
  preprocessor_line_error
  preprocessor_pragma
  image_macro
)

cts_api_test=(
  get_platform_info
  get_context_info
  get_device_info
  binary_get
  kernel_required_group_size
  release_kernel_order
  release_during_execute
  load_single_kernel
  load_two_kernels
  load_two_kernels_in_one
  load_two_kernels_manually
  get_program_info_kernel_names
  \# get_kernel_arg_info : too long
  create_kernels_in_program
  get_kernel_info
  kernel_private_memory_size
  execute_kernel_local_sizes
  set_kernel_arg_by_index
  set_kernel_arg_constant
  set_kernel_arg_struct_array
  kernel_global_constant
  \# kernel_attributes : too long
  min_max_thread_dimensions
  min_max_work_items_sizes
  min_max_work_group_size
  min_max_mem_alloc_size
  min_max_compute_units
  min_max_address_bits
  min_max_single_fp_config
  min_max_double_fp_config
  min_max_kernel_preferred_work_group_size_multiple
  min_max_language_version
  create_context_from_type
  platform_extensions
  get_platform_ids
  bool_type
  retain_queue_single
  retain_queue_multiple
  retain_mem_object_single
  retain_mem_object_multiple
  retain_mem_object_set_kernel_arg
  mem_object_destructor_callback
  null_buffer_arg
  negative_get_platform_info
  negative_get_platform_ids
)

Mifuns, 2023-10-26:
10.26更新后的data.sh

```
cts_basic_test=(
  hostptr           
  # fpmath              
  intmath_int         
  intmath_int2        
  intmath_int4        
  # intmath_long        
  # intmath_long2       
  # intmath_long4           
  # hiloeo              
  if                  
  # sizeof              
  loop                
  pointer_cast        
  # local_arg_def       
  # local_kernel_def    
  # local_kernel_scope  
  constant            
  constant_source     
  readimage           
  readimage_int16     
  readimage_fp32      
  writeimage          
  writeimage_int16    
  writeimage_fp32     
  mri_one             
  mri_multiple        
  image_r8            
  barrier             
  wg_barrier          
  # int2fp              
  # fp2int              
  imagereadwrite      
  imagereadwrite3d    
  readimage3d         
  readimage3d_int16   
  readimage3d_fp32    
  # bufferreadwriterect 
  arrayreadwrite      
  # arraycopy           
  imagearraycopy      
  imagearraycopy3d    
  imagecopy           
  imagecopy3d         
  imagerandomcopy     
  arrayimagecopy      
  arrayimagecopy3d    
  imagenpot           
  # vload_global        
  # vload_local         
  # vload_constant      
  # vload_private       
  # vstore_global       
  # vstore_local        
  # vstore_private      
  createkernelsinprogram  
  imagedim_pow2      
  imagedim_non_pow2   
  image_param        
  image_multipass_integer_coord 
  image_multipass_float_coord   
  # explicit_s2v_char   
  # explicit_s2v_uchar  
  # explicit_s2v_short  
  # explicit_s2v_ushort 
  # explicit_s2v_int    
  # explicit_s2v_uint   
  # explicit_s2v_long   
  # explicit_s2v_ulong  
  # explicit_s2v_float  
  explicit_s2v_double 
  # enqueue_map_buffer  
  enqueue_map_image   
  # work_item_functions 
  # astype              
  # async_copy_global_to_local          
  # async_copy_local_to_global          
  # async_strided_copy_global_to_local  
  # async_strided_copy_local_to_global  
  async_copy_global_to_local2D        
  async_copy_local_to_global2D        
  async_copy_global_to_local3D        
  async_copy_local_to_global3D        
  async_work_group_copy_fence_import_after_export_aliased_local               
  async_work_group_copy_fence_import_after_export_aliased_global              
  async_work_group_copy_fence_import_after_export_aliased_global_and_local    
  async_work_group_copy_fence_export_after_import_aliased_local               
  async_work_group_copy_fence_export_after_import_aliased_global              
  async_work_group_copy_fence_export_after_import_aliased_global_and_local    
  # prefetch                          
  kernel_call_kernel_function       
  host_numeric_constants            
  # kernel_numeric_constants          
  # kernel_limit_constants            
  # kernel_preprocessor_macros        
  # parameter_types                   
  # vector_creation                   
  # vector_swizzle                    
  # vec_type_hint                     
  # kernel_memory_alignment_local     
  kernel_memory_alignment_global    
  kernel_memory_alignment_constant  
  # kernel_memory_alignment_private   
  # progvar_prog_scope_misc           
  # progvar_prog_scope_uninit         
  # progvar_prog_scope_init           
  # progvar_func_scope                
  # global_work_offsets               
  get_global_offset                 
  global_linear_id                  
  # local_linear_id                   
  # enqueued_local_size               
  simple_read_image_pitch           
  simple_write_image_pitch          
  # get_linear_ids                    
  rw_image_access_qualifier         
)

cts_compiler_test=(
  load_program_source
  load_multistring_source
  load_two_kernel_source
  load_null_terminated_source
  load_null_terminated_multi_line_source
  load_null_terminated_partial_multi_line_source
  load_discreet_length_source
  get_program_source
  get_program_build_info
  get_program_info
  large_compile
  async_build
  options_build_optimizations
  preprocessor_define_udef
  preprocessor_include
  preprocessor_line_error
  preprocessor_pragma
  image_macro
)

cts_api_test=(
  get_platform_info
  get_context_info
  get_device_info
  binary_get
  kernel_required_group_size
  release_kernel_order
  release_during_execute
  load_single_kernel
  load_two_kernels
  load_two_kernels_in_one
  load_two_kernels_manually
  get_program_info_kernel_names
  # get_kernel_arg_info : too long
  create_kernels_in_program
  get_kernel_info
  kernel_private_memory_size
  execute_kernel_local_sizes
  set_kernel_arg_by_index
  set_kernel_arg_constant
  set_kernel_arg_struct_array
  kernel_global_constant
  # kernel_attributes : too long
  min_max_thread_dimensions
  min_max_work_items_sizes
  min_max_work_group_size
  min_max_mem_alloc_size
  min_max_compute_units
  min_max_address_bits
  min_max_single_fp_config
  min_max_double_fp_config
  min_max_kernel_preferred_work_group_size_multiple
  min_max_language_version
  create_context_from_type
  platform_extensions
  get_platform_ids
  bool_type
  retain_queue_single
  retain_queue_multiple
  retain_mem_object_single
  retain_mem_object_multiple
  retain_mem_object_set_kernel_arg
  mem_object_destructor_callback
  null_buffer_arg
  negative_get_platform_info
  negative_get_platform_ids
)

```

Mifuns, 2023-12-01:
12.1更新CTS除basic目录的其他目录已经可以通过的用例：
buffers:
```
buffer_read_async_int
buffer_read_async_uint
buffer_read_async_long
buffer_read_async_ulong
buffer_read_async_short
buffer_read_async_ushort
buffer_read_async_char
buffer_read_async_uchar
buffer_read_async_float
buffer_read_array_barrier_int
buffer_read_array_barrier_uint
buffer_read_array_barrier_long
buffer_read_array_barrier_ulong
buffer_read_array_barrier_short
buffer_read_array_barrier_ushort
buffer_read_array_barrier_char
buffer_read_array_barrier_uchar
buffer_read_array_barrier_float
buffer_read_int
buffer_read_uint
buffer_read_long
buffer_read_ulong
buffer_read_short
buffer_read_ushort
buffer_read_float
buffer_read_half
buffer_read_char
buffer_read_uchar
buffer_read_struct
buffer_read_random_size
buffer_map_read_int
buffer_map_read_uint
buffer_map_read_long
buffer_map_read_ulong
buffer_map_read_short
buffer_map_read_ushort
buffer_map_read_char
buffer_map_read_uchar
buffer_map_read_float
buffer_map_read_struct
buffer_map_write_int
buffer_map_write_uint
buffer_map_write_long
buffer_map_write_ulong
buffer_map_write_short
buffer_map_write_ushort
buffer_map_write_char
buffer_map_write_uchar
buffer_map_write_float
buffer_map_write_struct
buffer_write_int
buffer_write_uint
buffer_write_short
buffer_write_ushort
buffer_write_char
buffer_write_uchar
buffer_write_float
buffer_write_half
buffer_write_long
buffer_write_ulong
buffer_write_struct
buffer_write_async_int
buffer_write_async_uint
buffer_write_async_short
buffer_write_async_ushort
buffer_write_async_char
buffer_write_async_uchar
buffer_write_async_float
buffer_write_async_long
buffer_write_async_ulong
buffer_copy
buffer_partial_copy
mem_read_write_flags
mem_write_only_flags
mem_read_only_flags
mem_copy_host_flags
mem_alloc_ref_flags
array_info_size
sub_buffers_read_write_dual_devices
buffer_migrate
image_migrate
```

compiler:
```
load_program_source
load_multistring_source
load_two_kernel_source
load_null_terminated_source
load_null_terminated_multi_line_source
load_null_terminated_partial_multi_line_source
load_discreet_length_source
get_program_source
get_program_build_info
get_program_info
large_compile
async_build
options_build_optimizations
options_build_macro
options_build_macro_existence
options_include_directory
options_denorm_cache
preprocessor_define_udef
preprocessor_include
preprocessor_line_error
preprocessor_pragma
opencl_c_versions
image_macro
features_macro
unload_valid
```

events:
```
event_get_execute_status
event_get_write_array_status
event_get_read_array_status
event_get_info
event_wait_for_execute
event_wait_for_array
event_flush
event_finish_execute
event_finish_array
event_release_before_done
event_enqueue_marker
event_enqueue_marker_with_event_list
event_enqueue_barrier_with_event_list
out_of_order_event_waitlist_single_queue
out_of_order_event_waitlist_multi_queue
out_of_order_event_waitlist_multi_queue_multi_device
out_of_order_event_enqueue_wait_for_events_single_queue
out_of_order_event_enqueue_wait_for_events_multi_queue
out_of_order_event_enqueue_wait_for_events_multi_queue_multi_device
out_of_order_event_enqueue_marker_single_queue
out_of_order_event_enqueue_marker_multi_queue
out_of_order_event_enqueue_marker_multi_queue_multi_device
out_of_order_event_enqueue_barrier_single_queue
waitlists
```

extension:
全通过

generic_address_space:
`generic_ptr_to_host_mem_svm`

integer_ops:
```
integer_hadd
integer_rhadd
integer_mul_hi
integer_rotate
integer_max
integer_upsample
integer_abs
integer_abs_diff
integer_addAssign
integer_subtractAssign
integer_multiplyAssign
integer_divideAssign
integer_moduloAssign
integer_andAssign
integer_orAssign
integer_exclusiveOrAssign
extended_bit_ops_extract
extended_bit_ops_insert
extended_bit_ops_reverse
long_math
long_logic
long_shift
long_compare
ulong_math
ulong_logic
ulong_shift
ulong_compare
integer_dot_product
```

mem_host_flags:
全通过

multiple_device_context:
```
context_multiple_contexts_same_device
context_two_contexts_same_device
context_three_contexts_same_device
context_four_contexts_same_device
two_devices
```

pipes:
```
pipe_readwrite_long
pipe_readwrite_ulong
pipe_readwrite_half
pipe_workgroup_readwrite_long
pipe_workgroup_readwrite_ulong
pipe_workgroup_readwrite_half
pipe_subgroup_readwrite_int
pipe_subgroup_readwrite_uint
pipe_subgroup_readwrite_long
pipe_subgroup_readwrite_ulong
pipe_subgroup_readwrite_short
pipe_subgroup_readwrite_ushort
pipe_subgroup_readwrite_float
pipe_subgroup_readwrite_half
pipe_subgroup_readwrite_char
pipe_subgroup_readwrite_uchar
pipe_subgroup_readwrite_double
pipe_subgroup_readwrite_struct
pipe_convenience_readwrite_int
pipe_convenience_readwrite_long
pipe_convenience_readwrite_ulong
pipe_convenience_readwrite_half
pipe_subgroups_divergence
```

profiling:
全通过

relationals:
```
relational_any
relational_all
shuffle_copy
shuffle_function_call
shuffle_array_cast
shuffle_built_in
shuffle_built_in_dual_input
```

select:
```
select_half_ushort
select_half_short
select_ulong_ulong
select_ulong_long
select_long_ulong
select_long_long
select_double_ulong
select_double_long
```

subgroup:
全通过

SVM:
```
svm_byte_granularity
svm_set_kernel_exec_info_svm_ptrs
svm_fine_grain_memory_consistency
svm_fine_grain_sync_buffers
svm_shared_address_space_fine_grain
svm_shared_address_space_fine_grain_buffers
svm_migrate
```

vectors:
```
step_type
step_var
step_typedef_type
step_typedef_var
```

workgroups:
```
work_group_suggested_local_size_1D
work_group_suggested_local_size_2D
work_group_suggested_local_size_3D
```

wangqinfan, 2024-01-05:
2024.1.5更新CTS的basic目录下的通过情况：
```
cts_basic_test=(
  hostptr
  fpmath
  intmath_int
  intmath_int2
  intmath_int4
  intmath_long
  intmath_long2
  intmath_long4
  # hiloeo    #可以运行，时间巨长
  if
  sizeof
  loop
  pointer_cast
  local_arg_def
  # local_kernel_def
  local_kernel_scope
  constant
  constant_source
  readimage
  readimage_int16
  readimage_fp32
  writeimage
  writeimage_int16
  writeimage_fp32
  mri_one
  mri_multiple
  image_r8
  barrier
  wg_barrier
  int2fp  # only pass by setting parameter 32 
  fp2int
  imagereadwrite
  imagereadwrite3d
  readimage3d
  readimage3d_int16
  readimage3d_fp32
  bufferreadwriterect
  arrayreadwrite
  arraycopy
  imagearraycopy
  imagearraycopy3d
  imagecopy
  imagecopy3d
  imagerandomcopy
  arrayimagecopy
  arrayimagecopy3d
  imagenpot
  vload_global
  vload_local
  vload_constant
  vload_private
  vstore_global
  vstore_local  
  vstore_private
  createkernelsinprogram
  imagedim_pow2
  imagedim_non_pow2
  image_param
  image_multipass_integer_coord
  image_multipass_float_coord
  # explicit_s2v_char
  # explicit_s2v_uchar
  # explicit_s2v_short
  # explicit_s2v_ushort
  # explicit_s2v_int
  # explicit_s2v_uint
  # explicit_s2v_long
  # explicit_s2v_ulong
  # explicit_s2v_float
  # explicit_s2v_double
  # enqueue_map_buffer
  enqueue_map_image
  # work_item_functions
  # astype # pass but take too much time
  # async_copy_global_to_local
  # async_copy_local_to_global
  # async_strided_copy_global_to_local
  # async_strided_copy_local_to_global
  async_copy_global_to_local2D
  async_copy_local_to_global2D
  async_copy_global_to_local3D
  async_copy_local_to_global3D
  async_work_group_copy_fence_import_after_export_aliased_local
  async_work_group_copy_fence_import_after_export_aliased_global
  async_work_group_copy_fence_import_after_export_aliased_global_and_local
  async_work_group_copy_fence_export_after_import_aliased_local
  async_work_group_copy_fence_export_after_import_aliased_global
  async_work_group_copy_fence_export_after_import_aliased_global_and_local
  # prefetch
  kernel_call_kernel_function
  host_numeric_constants
  kernel_numeric_constants
  kernel_limit_constants
  kernel_preprocessor_macros
  parameter_types
  # vector_creation # pass but take too much time
  vector_swizzle
  vec_type_hint
  kernel_memory_alignment_local
  kernel_memory_alignment_global
  kernel_memory_alignment_constant
  kernel_memory_alignment_private
  # progvar_prog_scope_misc
  # progvar_prog_scope_uninit
  # progvar_prog_scope_init
  # progvar_func_scope
  global_work_offsets # only passed by no 32 parameter
  get_global_offset
  global_linear_id
  local_linear_id
  # enqueued_local_size
  simple_read_image_pitch
  simple_write_image_pitch
  get_linear_ids
  rw_image_access_qualifier
)
```

zhoujingya, 2024-03-22:
## build脚本更新
设置`VENTUS_INSTALL_PREFIX`与`cts_directory`
```
#!/usr/bin/bash
DIR=$(cd "$(dirname "${0}")" &> /dev/null && (pwd -W 2> /dev/null || pwd))
source ${DIR}/data.sh
# 指定要搜索的目录路径
# pocl_directory="/home/triton/work/OpenCL-CTS/TEST/pocl"
cts_directory="/home/triton/work/OpenCL-CTS/TEST/cts"
# 使用 find 命令列出目录层级下的所有文件
# pocl_executables=($(find "$pocl_directory" -type f -executable -print0 | xargs -0))
cts_executables=($(find "$cts_directory" -type f -executable -print0 | xargs -0))
#
export VENTUS_INSTALL_PREFIX="/home/triton/work/llvm-project/install"
export LD_LIBRARY_PATH=$VENTUS_INSTALL_PREFIX/lib
export OCL_ICD_VENDORS=$VENTUS_INSTALL_PREFIX/lib/libpocl.so
export POCL_DEVICES="ventus"

# # 遍历数组并执行每个可执行文件
# for file in "${pocl_executables[@]}"; do
#   echo "----"
#   if [ -x "$file" ]; then
#     echo "Running $file"
#     "$file" >& /dev/null
#     if [ $? -eq 0 ]; then
#       echo "$file execution successful"
#     else
#       echo "Execution failed with exit status $exit_status"
#       exit $?
#     fi
#   else
#     echo "$file is not executable or does not exist"
#   fi
# done

run_cts_basic() {

  for element in "${cts_basic_test[@]}"
    do
      echo "Running $1 $element"
      $1 $element >& /dev/null
      if [ $? -eq 0 ]; then
        echo "$1 $element execution successful"
      else
        echo "Execution failed with $1 $element"
        exit $?
      fi
    done
}

run_cts_compiler() {

  for element in "${cts_compiler_test[@]}"
    do
      echo "Running $1 $element"
      $1 $element >& /dev/null
      if [ $? -eq 0 ]; then
        echo "$1 $element execution successful"
      else
        echo "Execution failed with $1 $element"
        exit $?
      fi
    done
}

run_cts_api() {

  for element in "${cts_api_test[@]}"
    do
      echo "Running $1 $element"
      $1 $element >& /dev/null
      if [ $? -eq 0 ]; then
        echo "$1 $element execution successful"
      else
        echo "Execution failed with $1 $element"
      fi
    done
}

# CTS的测试不太一样，需要特别处理
for file in "${cts_executables[@]}"; do
  echo "----"
  if [[ "$file" == */test_basic ]]; then
    run_cts_basic $file
  elif [[ "$file" == */test_compiler ]]; then
    run_cts_compiler $file
  elif [[ "$file" == */test_api ]]; then
    run_cts_api $file
  else
    echo "$file is not executable or does not exist"
  fi
done

```

wangqinfan, 2024-03-28:
## 2024.3.28更新除basic, compiler, api外其他的部分目录通过情况（注释的是没通过的，有些通过的只是跳过了测试，会标注出来）：
### commonfns目录：
```
  clamp
  degrees
  fmax
  fmaxf
  fmin
  fminf
  max
  maxf
  min
  minf
  # mix // float16 failed, now double1 failed
  # mixf  // float16 failed, now double1 failed
  radians
  step
  stepf
  # smoothstep  //double1 fail
  # smoothstepf
  sign
```

### events目录：
`报：Device does not support out-of-order exec mode; skipping test.`

### extensions目录：
有5个子目录：
cl_ext_cxx_for_opencl目录报：Device does not support 'cl_ext_cxx_for_opencl'. Skipping the test.
cl_khr_command_buffer目录报：Device does not support 'cl_khr_command_buffer'. Skipping the test.
cl_khr_dx9_media_sharing目录没可执行文件
cl_khr_external_semaphore目录没可执行文件
cl_khr_semaphore目录报：cl_khr_semaphore is not supported on this platoform. Skipping test.

### generic_address_space目录：
没实现

### integer_ops目录：
```
  integer_clz
  integer_ctz
  integer_hadd
  integer_rhadd
  integer_mul_hi
  # integer_rotate
  integer_clamp 
  # integer_mad_sat # Vector uint16,uint16,uint16 FAILED
  # integer_mad_hi
  # integer_min
  # integer_max
  integer_upsample
  integer_abs
  integer_abs_diff
  integer_add_sat
  integer_sub_sat
  integer_addAssign
  integer_subtractAssign
  integer_multiplyAssign
  integer_divideAssign
  integer_moduloAssign
  integer_andAssign
  integer_orAssign
  integer_exclusiveOrAssign
  # unary_ops_increment
  unary_ops_decrement
  # unary_ops_full
  # integer_mul24
  # integer_mad24
  extended_bit_ops_extract # cl_khr_extended_bit_ops is not supported
  extended_bit_ops_insert # cl_khr_extended_bit_ops is not supported
[... 188 lines cut ...]
  execute # Note: device does not support images. Skipping test...
```

### relationals目录：
```
  relational_any
  relational_all
  relational_bitselect
  relational_select_signed
  relational_select_unsigned
  # relational_isequal # *** stack smashing detected ***: terminated
  # relational_isnotequal
  # relational_isgreater
  # relational_isgreaterequal
  # relational_isless
  # relational_islessequal
  # relational_islessgreater
  shuffle_copy
  shuffle_function_call
  shuffle_array_cast
  # shuffle_built_in
  shuffle_built_in_dual_input
```

未完

closed 2024-12-07 by zhoujingya