THU-DSP-LAB/llvm-project #29: [bug] 目前的资源使用量dump出来的信息是unknown

zhoujingya, 2023-09-25:
```
vecadd.riscv:   file format elf32-littleriscv

Disassembly of section .rodata.ventus.resource:

00000000 <.rodata.ventus.resource>:
       0: 00 00         <unknown>
       2: 00 00         <unknown>
       4: 60 00         <unknown>
       6: c8 01         <unknown>
```

zhoujingya, 2023-09-25:
```
/work/ventus-llvm/build/bin/llvm-readelf -x ".ventus.resource" vecadd.riscv
Hex dump of section '.ventus.resource':
0x00000000 00000000 00000000 00000000 00000000 ................
0x00000010 60000000 00000000 c8010000 00000000 `...............
```
姿势不对，但是得到的信息好像还是错的

zhoujingya, 2023-09-25:
```
.section	.ventus.resource,"w",@progbits
	.word	89
	.word	27
	.word	36
	.word	228
```

dodohack, 2023-09-26:
You can't expect objdump to work correctly for non .text sections.
This is correct behavior.

closed 2023-09-26 by dodohack