THU-DSP-LAB/llvm-project #56: [CTS] enqueued_local_size的问题

zhoujingya, 2023-11-03:
目前llvm提供的libclc函数里面缺少`enqueued_local_size`的`workitem`函数实现

zhoujingya, 2023-11-03:
初步调研发现，pocl里面主要是针对`global_size`能不能整除`local_size`的问题
[image]

closed 2024-05-31 by zhoujingya

Jules-Kong, 2024-07-31:
这个用例的实现是有限制条件的：

zhoujingya, 2024-07-31:
> 这个用例的实现是有限制条件的： lib\CL\pocl_ndrange_kernel.c ![image](https://private-user-images.githubusercontent.com/32475045/353682134-a5cd0803-dc3e-4295-a54d-88ab0dab58da.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjIzOTY5MTMsIm5iZiI6MTcyMjM5NjYxMywicGF0aCI6Ii8zMjQ3NTA0NS8zNTM2ODIxMzQtYTVjZDA4MDMtZGMzZS00Mjk1LWE1NGQtODhhYjBkYWI1OGRhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MzElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzMxVDAzMzAxM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkyNmRjODA2ODBhMzgwNTQxZTM2YTk0YjQwMGZjOWRmZmY0OTlhMGVjY2Q4OGQyMzQyY2ViNDZkMmZjYjUzMWQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.v-k-VVmooMP4dnfxwQymFCXmwtZzEbLRZbkZTrFrb0E)

是的呀