THU-DSP-LAB/llvm-project #74: [CTS] Invalid Object Idx

Jules-Kong, 2023-11-29:
- 子用例： **progvar_prog_scope_init**
- 报错信息：
[image]

zhoujingya, 2023-12-04:
这个测试的源文件是这样的

```
// A simple test for support of static variables inside a kernel.
int test_progvar_func_scope(cl_device_id device, cl_context context,
                            cl_command_queue queue, int num_elements)
{
    cl_bool skip{ CL_FALSE };
    auto error = should_skip(device, skip);
    if (CL_SUCCESS != error)
    {
        return TEST_FAIL;
    }
    if (skip)
    {
        log_info("Skipping progvar_func_scope since it is optionally not "
                 "supported on this device\n");
        return TEST_SKIPPED_ITSELF;
    }
...
...
...
```
`should_skip(device, skip)`这个函数的实现是这样的:
```
// Determines whether its valid to skip this test based on the driver version
// and the features it optionally supports.
// Whether the test should be skipped is writen into the out paramter skip.
// The check returns an error code for the clDeviceInfo query.
static cl_int should_skip(cl_device_id device, cl_bool& skip)
{
    // Assume we can't skip to begin with.
    skip = CL_FALSE;

    // Progvar tests are already skipped for OpenCL < 2.0, so here we only need
    // to test for 3.0 since that is when program scope global variables become
    // optional.
    if (get_device_cl_version(device) >= Version(3, 0))
    {
        size_t max_global_variable_size{};
        test_error(clGetDeviceInfo(device, CL_DEVICE_MAX_GLOBAL_VARIABLE_SIZE,
                                   sizeof(max_global_variable_size),
                                   &max_global_variable_size, nullptr),
                   "clGetDeviceInfo failed");
        skip = (max_global_variable_size != 0) ? CL_FALSE : CL_TRUE;
    }
    return CL_SUCCESS;
}
```
所以针对`ventus`这个设备，`get_device_cl_version`这个接口返回值有问题

zhoujingya, 2024-03-06:
其实这个issue严格来说可以放在pocl仓库 @Jules-Kong ,个人建议在pocl那边开一个，因为错误的原因我已经给你分析出来了

closed 2024-05-31 by zhoujingya

Jules-Kong, 2024-07-30:
> @zhoujingya get_device_cl_version函数返回值是ventus的版本2.0，那么skip依然是CL_FALSE，这存在什么问题？

[image]