THU-DSP-LAB/llvm-project #80: 函数调用的时候generic_addressing_space自动推导的问题

Mifuns, 2023-12-01:
OpenCL 2.0支持generic address space，所以在定义一个函数的时候可以不指定指针的地址空间，那么在调用该函数的时候可以传入任何地址空间的指针，所以函数内部不知道怎么翻译有关该指针的访存指令。可以看下面这个例子：

```
// RUN: clang -cl-std=CL2.0 -target riscv32 -mcpu=ventus-gpgpu -S %s
int test2(private int *a) {
    return *a + 1;
}

int test3(int *a) {
    return *a + 1;
}

int test1() {
    private int a[2];
    return test2(a) + test3(a) + test3(a);
}

```
这里test3和test2翻成了不同的访存指令。下面是汇编：
```
test2:
	addi	sp, sp, 4
	sw	ra, -4(sp)
	vlw.v	v0, 0(v0)
	vadd.vi	v0, v0, 1
	lw	ra, -4(sp)
	addi	sp, sp, -4
	ret
.Lfunc_end0:
	.size	test2, .Lfunc_end0-test2

	.globl	test3
	.p2align	2
	.type	test3,@function
test3:
	addi	sp, sp, 4
	sw	ra, -4(sp)
	vlw12.v	v0, 0(v0)
	vadd.vi	v0, v0, 1
	lw	ra, -4(sp)
	addi	sp, sp, -4
	ret
.Lfunc_end1:
	.size	test3, .Lfunc_end1-test3

	.globl	test1
	.p2align	2
	.type	test1,@function
test1:
	addi	sp, sp, 4
	sw	ra, -4(sp)
	lw	ra, -4(sp)
	addi	sp, sp, -4
	ret
```

Jules-Kong, 2024-02-22:
* 源码

* 命令
`clang -cl-std=CL2.0 -target riscv32 -mcpu=ventus-gpgpu -S -mllvm -print-after-all test_generic_addr_2.cl &> ventus_t2.log
clang -target amdgcn test_generic_addr_2.cl -mllvm -print-after-all -S &> amd_t2.log
`

* ventus与amd的差异
[image]

zhoujingya, 2024-02-22:
你编译AMDGPU target的时候下错了参数，`clang -target amdgcn test_generic_addr_2.cl -mllvm -print-after-all -S &> amd_t2.log`, 忘记加`-cl-std=CL2.0`了, @Jules-Kong