THU-DSP-LAB/llvm-project #107: 目前对于Divergent路径的分析存在问题

zhoujingya, 2024-03-25:
kernel函数的参数对于所有线程来说都是一样的,不应该是Divergent节点,参考`llvm/lib/Target/RISCV/RISCVTargetTransformInfo.cpp`

zhoujingya, 2024-03-26:
修改之后，编译会报"illegal copy from VGPR to SPGR",定位到原因出现在这里(目前的编译报错)
https://github.com/THU-DSP-LAB/llvm-project/blob/078bb26e2e8ab1d57ad9e55be42780ff794a948b/llvm/lib/CodeGen/SelectionDAG/InstrEmitter.cpp#L303-L350

zhoujingya, 2024-03-26:
https://github.com/THU-DSP-LAB/llvm-project/blob/078bb26e2e8ab1d57ad9e55be42780ff794a948b/llvm/lib/CodeGen/SelectionDAG/InstrEmitter.cpp#L326 这里有问题，分配的Register class不对

zhoujingya, 2024-03-26:
取消这里的注释 https://github.com/THU-DSP-LAB/llvm-project/blob/078bb26e2e8ab1d57ad9e55be42780ff794a948b/llvm/lib/Target/RISCV/RISCVTargetTransformInfo.cpp#L362-L364

编译以下代码成汇编代码：
```c
#pragma OPENCL EXTENSION cl_khr_fp64 : enable
__kernel void test_fn( __global double16 *srcValues, __global uint *offsets, __global double *destBuffer, uint alignmentOffset )
{
    int tid = get_global_id( 0 );
    vstore16( srcValues[ tid ], offsets[ tid ], destBuffer + alignmentOffset );
}

```
就可以复现报错

zhoujingya, 2024-03-26:
pin @wangqinfan @Jules-Kong

closed 2024-04-22 by zhoujingya

Jules-Kong, 2024-04-22:
No commit description and the simplest testcase.