THU-DSP-LAB/llvm-project #124: ld.lld: error undefined symbol: get_global_size(unsigned int)

suleymanemre, 2024-05-26:
Hello, I successfully installed the compiler designed for Ventus and got the results of the simple test I gave below.

```
__kernel void vector_add(__global const float *x, __global const float *y, __global float *restrict z) { // get index of the work item
        int index = get_global_id(0);
        // add the vector elements
         z[index] = x[index] + y[index];
}
```

After this test, I continued to try to get more test results, but the get_global_size and get_local_id functions I provided below produced error outputs.

### CONVOLUTION TEST
```
__kernel void Conv2D( __global int * image_in, //image input
                      __global int * filter_in, //filter input
                        int K, //filter kernel size
                     __global int * image_out) //feature map output
{
int W; //work group global size
int Wn; //padded image width
int x; //global id x
int y; //global id y
int ki, kj; //filter coordinate,(kj, ki)

int sum = 0; //multiply and sum of filter and data

x = get_global_id(0);
y = get_global_id(1);
W = get_global_size(0);
Wn = W + (K - 1);

for(ki=0; ki<K; ki++){
    for(kj=0; kj<K; kj++)
{
        sum  = sum + filter_in[ki*K + kj] * image_in[Wn*(y+ki) + x + kj];
}
}
image_out[y*W + x] = sum;
}
```

Error output of the above code:

```
suleyman@DESKTOP-UD9KE8O:~/Ventus_Compiler/llvm-project$ ./install/bin/clang -cl-std=CL3.0 -target riscv32 -mcpu=ventus-gpgpu convolution.cl  ./install/lib/crt0.o -L./install/lib -lworkitem -I./libclc/generic/include -nodefaultlibs ./
libclc/riscv32/lib/workitem/get_global_id.cl -O1 -cl-std=CL2.0 -Wl,-T,utils/ldscripts/ventus/elf32lriscv.ld -o convol
ution.riscv
ld.lld: error: undefined symbol: get_global_size(unsigned int)
>>> referenced by convolution.cl
>>>               /tmp/convolution-94b5ec.o:(Conv2D)
clang-16: error: ld.lld command failed with exit code 1 (use -v to see invocation)
```

###  ADD NUMBER TEST

```
__kernel void add_numbers(__global float4* data,
                                         __local float* local_result, __global float* group_result) {

   float sum;
   float4 input1, input2, sum_vector; // array of 4 floats which support vectorization
   uint global_addr, local_addr;

   global_addr = get_global_id(0) * 2;
   input1 = data[global_addr];
   input2 = data[global_addr+1];
   sum_vector = input1 + input2; // perform four floating-point additions simultaneously

   local_addr = get_local_id(0);
   local_result[local_addr] = sum_vector.s0 + sum_vector.s1 +
                              sum_vector.s2 + sum_vector.s3;
   barrier(CLK_LOCAL_MEM_FENCE);

   if(get_local_id(0) == 0) {
      sum = 0.0f;
      for(int i=0; i<get_local_size(0); i++) {
                       sum += local_result[i];
        }
     group_result[get_group_id(0)] = sum;
 }
 }
```

Error output of the above code:
```
suleyman@DESKTOP-UD9KE8O:~/Ventus_Compiler/llvm-project$ ./install/bin/clang -cl-std=CL3.0 -target riscv32 -mcpu=ventus-gpgpu add_numbers.cl  ./install/lib/crt0.o -L./install/lib -lworkitem -I./libclc/generic/include -nodefaultlibs ./libclc/riscv32/lib/workitem/get_global_id.cl -O1 -cl-std=CL2.0 -Wl,-T,utils/ldscripts/ventus/elf32lriscv.ld -o convolution.riscv
ld.lld: error: undefined symbol: get_local_id(unsigned int)
>>> referenced by add_numbers.cl
>>>               /tmp/add_numbers-cc403c.o:(add_numbers)

ld.lld: error: undefined symbol: get_local_size(unsigned int)
>>> referenced by add_numbers.cl
>>>               /tmp/add_numbers-cc403c.o:(add_numbers)

ld.lld: error: undefined symbol: get_group_id(unsigned int)
>>> referenced by add_numbers.cl
>>>               /tmp/add_numbers-cc403c.o:(add_numbers)
clang-16: error: ld.lld command failed with exit code 1 (use -v to see invocation)
```

How can i solve this problem.

zhoujingya, 2024-05-27:
add header file like below
```
#include <clc.clc.h>
```
besides, -cl-std=CL2.0, ventus dose not support OpenCL3.0 by now @suleymanemre

suleymanemre, 2024-05-29:
Thank you for your time and answer, I'm a little busy right now but I'll write when I try. @zhoujingya

closed 2024-06-05 by zhoujingya