THU-DSP-LAB/llvm-project #125: 数据拷贝相关程序编译问题

xxrrnn, 2024-05-27:
kernel函数如下：
```
__kernel void __attribute__((noinline)) lsu(__global int *src)
{
  // long i = get_global_id(0);
  // dst[i] += src[i];
  __local int sharedmem[32];
  for(int i = 0; i < 32; i++){
    sharedmem[i] = src[i];
  }
}
```
main.cc函数如下：
```
/*
 *  Simple OpenCL demo program
 *
 *  Copyright (C) 2009  Clifford Wolf <clifford@clifford.at>
 *
 *  This program is free software; you can redistribute it and/or modify
 *  it under the terms of the GNU General Public License as published by
 *  the Free Software Foundation; either version 2 of the License, or
 *  (at your option) any later version.
 *
 *  This program is distributed in the hope that it will be useful,
 *  but WITHOUT ANY WARRANTY; without even the implied warranty of
 *  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 *  GNU General Public License for more details.
 *
 *  You should have received a copy of the GNU General Public License
 *  along with this program; if not, write to the Free Software
 *  Foundation, Inc., 59 Temple Place, Suite 330, Boston, MA  02111-1307  USA
 *
 *  gcc -o cldemo -std=gnu99 -Wall -I/usr/include/nvidia-current cldemo.c
 * -lOpenCL
 *
 */

#include <CL/cl.h>
#include <fstream>
#include <iostream>
#include <sstream>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <unistd.h> 
#include <chrono>
#include <vector>

#define CL_CHECK(_expr)                                                        \
  do {                                                                         \
    cl_int _err = _expr;                                                       \
    if (_err == CL_SUCCESS)                                                    \
      break;                                                                   \
    fprintf(stderr, "OpenCL Error: '%s' returned %d!\n", #_expr, (int)_err);   \
    abort();                                                                   \
  } while (0)

#define CL_CHECK_ERR(_expr)                                                    \
  ({                                                                           \
    cl_int _err = CL_INVALID_VALUE;                                            \
    decltype(_expr) _ret = _expr;                                                \
    if (_err != CL_SUCCESS) {                                                  \
      fprintf(stderr, "OpenCL Error: '%s' returned %d!\n", #_expr, (int)_err); \
      abort();                                                                 \
    }                                                                          \
    _ret;                                                                      \
  })

void pfn_notify(const char *errinfo, const void *private_info, size_t cb,
[... 231 lines cut ...]
.PHONY: lsu.out
lsu.out: 
	$(CC) $(CC_FLAGS) main.cc -o lsu.out -I$(OPENCL_INC) -L$(OPENCL_LIB) -lOpenCL -Wno-unused-result -Wl,--init=lsu

%.o: %.[ch]
	$(CC) $(CC_FLAGS) $< -c

clean:
	rm -f *.o *~ *.out *.linkinfo
```

通过mask和执行./lsu.out，生成的dump文件中没有kernel中的指令(以下为dump文件的一部分)：
```
800000ac <.Lpcrel_hi6>:
800000ac: 97 12 00 00  	auipc	t0, 1
800000b0: 93 82 42 f5  	addi	t0, t0, -172
800000b4: 23 a0 62 00  	sw	t1, 0(t0)

800000b8 <lsu>:
800000b8: 67 80 00 00  	ret

```

[kernel.txt](https://github.com/THU-DSP-LAB/llvm-project/files/15456281/kernel.txt)
[main.txt](https://github.com/THU-DSP-LAB/llvm-project/files/15456282/main.txt)

zhoujingya, 2024-05-27:
`include ../common/make.config`这里面是啥,你们这边写的user mode driver code,生成的是host(X86)的代码@xxrrnn

xxrrnn, 2024-05-28:
# DEFAULT OCL
OPENCL_DIR = /home/lan/work/ventus/ocl-icd
OPENCL_INC = /home/lan/work/ventus/llvm-project/install/include
OPENCL_LIB = /home/lan/work/ventus/llvm-project/install/lib

zhoujingya, 2024-05-28:
> # DEFAULT OCL
> OPENCL_DIR = /home/lan/work/ventus/ocl-icd OPENCL_INC = /home/lan/work/ventus/llvm-project/install/include OPENCL_LIB = /home/lan/work/ventus/llvm-project/install/lib

需要调用kernel的driver(pocl)来编译kernel代码。因为你现在这种方式就是user mode，类似于pocl里面的vecadd例子，编译的是host代码，

xxrrnn, 2024-05-28:
具体该怎么做呢？还不是很熟悉

zhoujingya, 2024-05-28:
> 具体该怎么做呢？还不是很熟悉

@yangzexia 如果你们在一个地方的话可以当面问一下

xxrrnn, 2024-05-28:
```
lan@DESKTOPONFR7BU:~/work/ventus/gpu-rodinia/vortexcase/lsu$ make
clang++ -g -O2 -std=c++11 -I. main.cc -o lsu.out -I/home/lan/work/ventus/llvm-project/install/include -L/home/lan/work/ventus/llvm-project/install/lib -lOpenCL -Wno-unused-result -Wl,--init=lsu
In file included from main.cc:25:
In file included from /home/lan/work/ventus/llvm-project/install/include/CL/cl.h:20:
/home/lan/work/ventus/llvm-project/install/include/CL/cl_version.h:22:9: warning: cl_version.h: CL_TARGET_OPENCL_VERSION is not defined. Defaulting to 300 (OpenCL 3.0) [-W#pragma-messages]
#pragma message("cl_version.h: CL_TARGET_OPENCL_VERSION is not defined. Defaulting to 300 (OpenCL 3.0)")
        ^
main.cc:166:24: warning: 'clCreateCommandQueue' is deprecated [-Wdeprecated-declarations]
  queue = CL_CHECK_ERR(clCreateCommandQueue(context, device_id, 0, &_err));
                       ^
/home/lan/work/ventus/llvm-project/install/include/CL/cl.h:1923:66: note: 'clCreateCommandQueue' has been explicitly marked deprecated here
                     cl_int *                       errcode_ret) CL_API_SUFFIX__VERSION_1_2_DEPRECATED;
                                                                 ^
/home/lan/work/ventus/llvm-project/install/include/CL/cl_platform.h:108:72: note: expanded from macro 'CL_API_SUFFIX__VERSION_1_2_DEPRECATED'
    #define CL_API_SUFFIX__VERSION_1_2_DEPRECATED CL_API_SUFFIX_COMMON CL_API_SUFFIX_DEPRECATED
                                                                       ^
/home/lan/work/ventus/llvm-project/install/include/CL/cl_platform.h:78:51: note: expanded from macro 'CL_API_SUFFIX_DEPRECATED'
  #define CL_API_SUFFIX_DEPRECATED __attribute__((deprecated))
                                                  ^
main.cc:166:24: warning: 'clCreateCommandQueue' is deprecated [-Wdeprecated-declarations]
  queue = CL_CHECK_ERR(clCreateCommandQueue(context, device_id, 0, &_err));
                       ^
/home/lan/work/ventus/llvm-project/install/include/CL/cl.h:1923:66: note: 'clCreateCommandQueue' has been explicitly marked deprecated here
                     cl_int *                       errcode_ret) CL_API_SUFFIX__VERSION_1_2_DEPRECATED;
                                                                 ^
/home/lan/work/ventus/llvm-project/install/include/CL/cl_platform.h:108:72: note: expanded from macro 'CL_API_SUFFIX__VERSION_1_2_DEPRECATED'
    #define CL_API_SUFFIX__VERSION_1_2_DEPRECATED CL_API_SUFFIX_COMMON CL_API_SUFFIX_DEPRECATED
                                                                       ^
/home/lan/work/ventus/llvm-project/install/include/CL/cl_platform.h:78:51: note: expanded from macro 'CL_API_SUFFIX_DEPRECATED'
  #define CL_API_SUFFIX_DEPRECATED __attribute__((deprecated))
                                                  ^
3 warnings generated.
lan@DESKTOPONFR7BU:~/work/ventus/gpu-rodinia/vortexcase/lsu$ ./lsu.out 
Workload size=64
** Final POCL_DEBUG flags: 100000 
[INFO]: [HW DRIVER] in [FILE] ventus.cpp,[LINE]25,[fn] vt_dev_open: vt_dev_open : hello world from ventus.cpp
spike device initialize: allocating local memory: to allocate at 0x70000000 with 268435456 bytes 
spike device initialize: allocating pc source memory: to allocate at 0x80000000 with 268435456 bytes 
Create program from kernel source
### options: -DPOCL_DEVICE_ADDRESS_BITS=32 -D__USE_CLANG_OPENCL_C_H -xcl -Dinline= -I. -cl-kernel-arg-info  -D__ENDIAN_LITTLE__=1 -DCL_DEVICE_MAX_GLOBAL_VARIABLE_SIZE=0 -D__OPENCL_VERSION__=200 -cl-std=CL2.0 -D__OPENCL_C_VERSION__=200 -Dcl_khr_fp64=1 -D__opencl_c_generic_address_space=1 -D__opencl_c_named_address_space_builtins=1 -cl-ext=-all,+cl_khr_fp64,+__opencl_c_generic_address_space,+__opencl_c_named_address_space_builtins -fno-builtin -triple=riscv32 -target-cpu ventus-gpgpu user_options: 
### Triple: riscv32, CPU: ventus-gpgpu
create input buffer
create output buffer
create kernel
setting up kernel args
initialize buffers
Warning: the memory at  [0x90000000, 0x900000FF] has been realigned
to the 4 KiB page size: [0x90000000, 0x90000FFF]
to allocate at 0x90000000 with 4096 bytes 
to copy to 0x90000000 with 256 bytes
enqueue kernel
notice that ventus hasn't support local buffer as argument yet.
[2024-05-28 03:28:56.184955456]POCL: in fn void pocl_ventus_run(void *, _cl_command_node *) at line 629:
  |    VENTUS |  Allocating kernel arg buffer entry:
Warning: the memory at  [0x90001000, 0x90001003] has been realigned
to the 4 KiB page size: [0x90001000, 0x90001FFF]
to allocate at 0x90001000 with 4096 bytes 
to copy to 0x90001000 with 4 bytes
[2024-05-28 03:28:56.140411276613688]POCL: in fn void pocl_ventus_run(void *, _cl_command_node *) at line 683:
  |    VENTUS |  Kernel entry of "lsu" is : "0x800000b8"
[2024-05-28 03:28:56.345683756]POCL: in fn void pocl_ventus_run(void *, _cl_command_node *) at line 714:
  |    VENTUS |  Vmem file has been written to object.vmem
[2024-05-28 03:28:56.345863956]POCL: in fn void pocl_ventus_run(void *, _cl_command_node *) at line 766:
  |    VENTUS |  Preparing private memory of ventus:
to allocate at 0x90002000 with 262144 bytes 
[2024-05-28 03:28:56.345930556]POCL: in fn void pocl_ventus_run(void *, _cl_command_node *) at line 801:
  |    VENTUS |  Allocating metadata space:
Warning: the memory at  [0x90042000, 0x9004203F] has been realigned
to the 4 KiB page size: [0x90042000, 0x90042FFF]
to allocate at 0x90042000 with 4096 bytes 
to copy to 0x90042000 with 64 bytes
[... cut ...]
arg gpgpu is numw:4,numt:16,numwg:1,kernelx:1,kernely:1,kernelz:1,ldssize:0x1000,pdssize:0x10000000,pdsbase:0x90002000,knlbase:0x90042000,currwgid:0
vaddr mem scope is -m0x70000000:0x90043000
src file is object.riscv, run log is written to object.riscv.log
spike -l --log-commits -p4 --isa rv32gcv_zfh --pc=0x80000000 -m0x70000000:0x90043000 --varch vlen:512,elen:32 --gpgpuarch numw:4,numt:16,numwg:1,kernelx:1,kernely:1,kernelz:1,ldssize:0x1000,pdssize:0x10000000,pdsbase:0x90002000,knlbase:0x90042000,currwgid:0 --log=object.riscv.log object.riscv 
Log file object.riscv.log renamed successfully to lsu_0.log.
Elapsed time: 170 ms
```

zhoujingya, 2024-05-28:
> ```
> lan@DESKTOPONFR7BU:~/work/ventus/gpu-rodinia/vortexcase/lsu$ make
> clang++ -g -O2 -std=c++11 -I. main.cc -o lsu.out -I/home/lan/work/ventus/llvm-project/install/include [...]

这样看着过程是对的，有什么问题吗？

yangzexia, 2024-05-28:
kernel的编译选项：
[image]

zhoujingya, 2024-05-28:
> kernel的编译选项： [...]

看着没啥毛病啊

zhoujingya, 2024-05-28:
[lsu.log](https://pastebin.ubuntu.com/p/V4BXZsR8Gs/)
[vecadd.log](https://pastebin.ubuntu.com/p/NyhjGtzDgR/)
感觉像是你们的c++代码写的有问题 @yangzexia @xxrrnn

zhoujingya, 2024-05-28:
代码被优化掉了，此issue close

closed 2024-05-28 by zhoujingya