THU-DSP-LAB/llvm-project #129: 执行build-ventus.sh出错

chiliao, 2024-06-24:
[image]

按照readme拉下来了所有的库，所有的仓库都是在主分支，执行bash build-ventus.sh，在编译libclc时报错：
[10/14] Building CLC object CMakeFiles/builtins.link.riscv32--.dir/generic/lib/math/clc_tan.bc
FAILED: CMakeFiles/builtins.link.riscv32--.dir/generic/lib/math/clc_tan.bc
/home/dpu/liumin/ventus/llvm-project/install/bin/clang -DCLC_RISCV32 -D__CLC_INTERNAL -I/home/dpu/liumin/ventus/llvm-project/install/include -I/home/dpu/liumin/ventus/llvm-project/libclc/generic/include -target riscv32 -mcpu=ventus-gpgpu -cl-std=CL2.0 -I/home/dpu/liumin/ventus/llvm-project/libclc/generic/include -Dcl_khr_fp64  -ffunction-sections -fdata-sections -cl-no-stdinc -target riscv32-- -fno-builtin -nostdlib -I /home/dpu/liumin/ventus/llvm-project/libclc/generic/lib/math -o CMakeFiles/builtins.link.riscv32--.dir/generic/lib/math/clc_tan.bc -c /home/dpu/liumin/ventus/llvm-project/libclc/generic/lib/math/clc_tan.cl -emit-llvm && /home/dpu/liumin/ventus/llvm-project/install/bin/clang -DCLC_RISCV32 -D__CLC_INTERNAL -I/home/dpu/liumin/ventus/llvm-project/install/include -I/home/dpu/liumin/ventus/llvm-project/libclc/generic/include -target riscv32 -mcpu=ventus-gpgpu -cl-std=CL2.0 -I/home/dpu/liumin/ventus/llvm-project/libclc/generic/include -Dcl_khr_fp64  -ffunction-sections -fdata-sections -cl-no-stdinc -target riscv32-- -fno-builtin -nostdlib -I /home/dpu/liumin/ventus/llvm-project/libclc/generic/lib/math -o CMakeFiles/builtins.link.riscv32--.dir/generic/lib/math/clc_tan.bc.o -c /home/dpu/liumin/ventus/llvm-project/libclc/generic/lib/math/clc_tan.cl
clang: /home/dpu/liumin/ventus/llvm-project/llvm/lib/Target/RISCV/RISCVInstrInfo.cpp:143: unsigned int llvm::RISCVInstrInfo::getPrivateMemoryOpcode(llvm::MachineInstr&) const: Assertion `0 && "TODO"' failed.
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace, preprocessed source, and associated run script.
Stack dump:
0.      Program arguments: /home/dpu/liumin/ventus/llvm-project/install/bin/clang -DCLC_RISCV32 -D__CLC_INTERNAL -I/home/dpu/liumin/ventus/llvm-project/install/include -I/home/dpu/liumin/ventus/llvm-project/libclc/generic/include -target riscv32 -mcpu=ventus-gpgpu -cl-std=CL2.0 -I/home/dpu/liumin/ventus/llvm-project/libclc/generic/include -Dcl_khr_fp64 -ffunction-sections -fdata-sections -cl-no-stdinc -target riscv32-- -fno-builtin -nostdlib -I /home/dpu/liumin/ventus/llvm-project/libclc/generic/lib/math -o CMakeFiles/builtins.link.riscv32--.dir/generic/lib/math/clc_tan.bc.o -c /home/dpu/liumin/ventus/llvm-project/libclc/generic/lib/math/clc_tan.cl
1.      <eof> parser at end of file
2.      Code generation
3.      Running pass 'Function Pass Manager' on module '/home/dpu/liumin/ventus/llvm-project/libclc/generic/lib/math/clc_tan.cl'.
4.      Running pass 'Prologue/Epilogue Insertion & Frame Finalization' on function '@_Z9__clc_tanf'
 #0 0x00007f8c385b025f PrintStackTraceSignalHandler(void*) Signals.cpp:0:0
 #1 0x00007f8c385add84 llvm::sys::CleanupOnSignal(unsigned long) (/home/dpu/liumin/ventus/llvm-project/install/lib/libLLVMSupport.so.16git+0x20dd84)
 #2 0x00007f8c3847d320 CrashRecoverySignalHandler(int) CrashRecoveryContext.cpp:0:0
 #3 0x00007f8c37c64f10 (/lib/x86_64-linux-gnu/libc.so.6+0x3ef10)
 #4 0x00007f8c37c64e87 raise /build/glibc-CVJwZb/glibc-2.27/signal/../sysdeps/unix/sysv/linux/raise.c:51:0
 #5 0x00007f8c37c667f1 abort /build/glibc-CVJwZb/glibc-2.27/stdlib/abort.c:81:0
 #6 0x00007f8c37c563fa __assert_fail_base /build/glibc-CVJwZb/glibc-2.27/assert/assert.c:89:0
 #7 0x00007f8c37c56472 (/lib/x86_64-linux-gnu/libc.so.6+0x30472)
 #8 0x00007f8c40272639 llvm::RISCVInstrInfo::getPrivateMemoryOpcode(llvm::MachineInstr&) const (/home/dpu/liumin/ventus/llvm-project/install/lib/libLLVMRISCVCodeGen.so.16git+0x5c639)
 #9 0x00007f8c402f977f llvm::RISCVRegisterInfo::eliminateFrameIndex(llvm::MachineInstrBundleIterator<llvm::MachineInstr, false>, int, unsigned int, llvm::RegScavenger*) const (/home/dpu/liumin/ventus/llvm-project/install/lib/libLLVMRISCVCodeGen.so.16git+0xe377f)
#10 0x00007f8c3db99d82 (anonymous namespace)::PEI::replaceFrameIndices(llvm::MachineBasicBlock*, llvm::MachineFunction&, int&) PrologEpilogInserter.cpp:0:0
#11 0x00007f8c3db9ee71 (anonymous namespace)::PEI::runOnMachineFunction(llvm::MachineFunction&) PrologEpilogInserter.cpp:0:0
#12 0x00007f8c3da39c82 llvm::MachineFunctionPass::runOnFunction(llvm::Function&) (.part.71) MachineFunctionPass.cpp:0:0
#13 0x00007f8c39565ea6 llvm::FPPassManager::runOnFunction(llvm::Function&) (/home/dpu/liumin/ventus/llvm-project/install/lib/libLLVMCore.so.16git+0x24cea6)
#14 0x00007f8c395661f9 llvm::FPPassManager::runOnModule(llvm::Module&) (/home/dpu/liumin/ventus/llvm-project/install/lib/libLLVMCore.so.16git+0x24d1f9)
#15 0x00007f8c39567070 llvm::legacy::PassManagerImpl::run(llvm::Module&) (/home/dpu/liumin/ventus/llvm-project/install/lib/libLLVMCore.so.16git+0x24e070)
#16 0x00007f8c3e41d28e clang::EmitBackendOutput(clang::DiagnosticsEngine&, clang::HeaderSearchOptions const&, clang::CodeGenOptions const&, clang::TargetOptions const&, clang::LangOptions const&, llvm::StringRef, llvm::Module*, clang::BackendAction, std::unique_ptr<llvm::raw_pwrite_stream, std::default_delete<llvm::raw_pwrite_stream>>) (/home/dpu/liumin/ventus/llvm-project/install/lib/libclangCodeGen.so.16git+0xeb28e)
#17 0x00007f8c3e83799b clang::BackendConsumer::HandleTranslationUnit(clang::ASTContext&) (/home/dpu/liumin/ventus/llvm-project/install/lib/libclangCodeGen.so.16git+0x50599b)
#18 0x00007f8c3167df09 clang::ParseAST(clang::Sema&, bool, bool) (/home/dpu/liumin/ventus/llvm-project/install/lib/libclangParse.so.16git+0x34f09)
#19 0x00007f8c3e836658 clang::CodeGenAction::ExecuteAction() (/home/dpu/liumin/ventus/llvm-project/install/lib/libclangCodeGen.so.16git+0x504658)
#20 0x00007f8c3c735f49 clang::FrontendAction::Execute() (/home/dpu/liumin/ventus/llvm-project/install/lib/libclangFrontend.so.16git+0x12af49)
#21 0x00007f8c3c6c382a clang::CompilerInstance::ExecuteAction(clang::FrontendAction&) (/home/dpu/liumin/ventus/llvm-project/install/lib/libclangFrontend.so.16git+0xb882a)
#22 0x00007f8c3fdd1513 clang::ExecuteCompilerInvocation(clang::CompilerInstance*) (/home/dpu/liumin/ventus/llvm-project/install/lib/libclangFrontendTool.so.16git+0x4513)
#23 0x000056146daa5cd4 cc1_main(llvm::ArrayRef<char const*>, char const*, void*) (/home/dpu/liumin/ventus/llvm-project/install/bin/clang+0x16cd4)
#24 0x000056146daa13e8 ExecuteCC1Tool(llvm::SmallVectorImpl<char const*>&) driver.cpp:0:0
#25 0x00007f8c3c1d25d5 void llvm::function_ref<void ()>::callback_fn<clang::driver::CC1Command::Execute(llvm::ArrayRef<std::optional<llvm::StringRef>>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>>*, bool*) const::'lambda'()>(long) Job.cpp:0:0
#26 0x00007f8c3847dab3 llvm::CrashRecoveryContext::RunSafely(llvm::function_ref<void ()>) (/home/dpu/liumin/ventus/llvm-project/install/lib/libLLVMSupport.so.16git+0xddab3)
#27 0x00007f8c3c1d54de clang::driver::CC1Command::Execute(llvm::ArrayRef<std::optional<llvm::StringRef>>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>>*, bool*) const (/home/dpu/liumin/ventus/llvm-project/install/lib/libclangDriver.so.16git+0xbd4de)
#28 0x00007f8c3c19e393 clang::driver::Compilation::ExecuteCommand(clang::driver::Command const&, clang::driver::Command const*&, bool) const (/home/dpu/liumin/ventus/llvm-project/install/lib/libclangDriver.so.16git+0x86393)
#29 0x00007f8c3c19ef83 clang::driver::Compilation::ExecuteJobs(clang::driver::JobList const&, llvm::SmallVectorImpl<std::pair<int, clang::driver::Command const*>>&, bool) const (/home/dpu/liumin/ventus/llvm-project/install/lib/libclangDriver.so.16git+0x86f83)
#30 0x00007f8c3c1ab5bc clang::driver::Driver::ExecuteCompilation(clang::driver::Compilation&, llvm::SmallVectorImpl<std::pair<int, clang::driver::Command const*>>&) (/home/dpu/liumin/ventus/llvm-project/install/lib/libclangDriver.so.16git+0x935bc)
#31 0x000056146daa3f9a clang_main(int, char**) (/home/dpu/liumin/ventus/llvm-project/install/bin/clang+0x14f9a)
#32 0x00007f8c37c47c87 __libc_start_main /build/glibc-CVJwZb/glibc-2.27/csu/../csu/libc-start.c:344:0
#33 0x000056146da9c18a _start (/home/dpu/liumin/ventus/llvm-project/install/bin/clang+0xd18a)
clang-16: error: clang frontend command failed with exit code 134 (use -v to see invocation)
clang version 16.0.0
Target: riscv32-unknown-unknown
Thread model: posix
InstalledDir: /home/dpu/liumin/ventus/llvm-project/install/bin
clang-16: note: diagnostic msg:
********************

PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
Preprocessed source(s) and associated run script(s) are located at:
clang-16: note: diagnostic msg: /tmp/clc_tan-c3e9f9.cl
clang-16: note: diagnostic msg: /tmp/clc_tan-c3e9f9.sh
clang-16: note: diagnostic msg:
请问这个应该怎么解决呢？

zhoujingya, 2024-06-24:
@chiliao 我看看

zhoujingya, 2024-06-24:
@chiliao #130 你要是急的话先打这个补丁试试？

chiliao, 2024-06-25:
@zhoujingya 好的，我试试.

chiliao, 2024-06-25:
@zhoujingya 编译通过了，谢谢。

closed 2024-06-25 by zhoujingya