THU-DSP-LAB/llvm-project #136: if分支中sGPR被修改导致else分支计算错误

ziliangzl, 2024-07-08:
在执行CTS的geometrics test_geometrics geom_cross测试中，计算结果错误，定位到问题出现在`libclc/riscv32/lib/compiler-rt/adddf3.cl`的`__addXf3__`函数中。
执行以下命令并查看CFG图：
```sh
./build/bin/clang -target riscv32 -mcpu=ventus-gpgpu -S -emit-llvm libclc/riscv32/lib/compiler-rt/adddf3.cl
./build/bin/opt -S -print-after='Insert-join-to-VBranch' adddf3.ll -o adddf3-after.ll
./build/bin/opt -p dot-cfg adddf3-after.ll
```
[image]
如上图所示，在if else分支中存在提前return的逻辑，导致PostIDomBB为`cleanup`基本块，join指令被插入在`cleanup`中。
在`adddf3-after.ll`中，一个分支执行之后会从endif一直执行到cleanup才join到另一个分支，其中x5寄存器的值会被修改，导致计算出错。
在endif中，使用到了x5寄存器:
```
bb.84.if.end117:
...
renamable $x5 = ADD killed renamable $x5, killed renamable $x6
...
$v6 = VMV_V_X killed $x5
...
```
在endif和cleanup中间的代码中，修改了x5寄存器的值：
```
bb.85.if.then120:
...
  renamable $x5 = LUI 524032
```
导致第二次另一个分支执行到endif时，x5的值不是想要的值。

zhoujingya, 2024-07-08:
@ziliangzl 我记得好像写过一个分析插入VMV_V_X的代码， 也在`llvm/lib/Target/RISCV/VentusInsertJoinToVBranch.cpp`这个文件里面，可以先看一下？

zhoujingya, 2024-08-16:
Fixed by #139

closed 2024-08-16 by zhoujingya