THU-DSP-LAB/llvm-project #145: [ventus][cts] flw/fsw 指令的处理引起新的cts/rellations用例错误

Jules-Kong, 2024-08-16:
* 简述
  原本cts/rellations的子用例是全部通过的，但是llvm的#112的合入导致下面用例的失败：
shuffle_copy、shuffle_function_call、shuffle_array_cast、shuffle_built_in_dual_input；
* 报错情况
 类似于
[image]

zhoujingya, 2024-08-16:
> * 简述
>   原本cts/rellations的子用例是全部通过的，但是llvm的#112的合入导致下面用例的失败：
>   shuffle_copy、shuffle_function_call、shuffle_array_cast、shuffle_built_in_dual_input；
> * 报错情况
>   类似于
>   ![Screenshot from 2024-08-16 17-15-24](https://private-user-images.githubusercontent.com/32475045/358534174-b96f3eaa-b389-4ab0-8288-5e1388c2606a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM4MDE2MjQsIm5iZiI6MTcyMzgwMTMyNCwicGF0aCI6Ii8zMjQ3NTA0NS8zNTg1MzQxNzQtYjk2ZjNlYWEtYjM4OS00YWIwLTgyODgtNWUxMzg4YzI2MDZhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODE2VDA5NDIwNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY2OGQ3ODNmZDMxMGJiNWRkYTY0ZmY1NmExZTM1YzAyYjhmNjNmZmY3ODg0Nzc2NmJlODYxYzQzMWEzMGI5NjcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.d52YlqPt5MZy0aFa16b99fMTgrEuWFzjDF0jJFEng4Y)

我比较好奇的是这个是什么软件测试完得到的图片?

zhoujingya, 2024-08-18:
测试环境？都是基于所有仓库main分支？

closed 2024-08-21 by zhoujingya