THU-DSP-LAB/llvm-project #150: vload3/vstore3函数链接失败

Jules-Kong, 2024-10-08:
- 报错：

[image]

[image]

- 涉及到的用例：
CTS-geometrics：geom_length、geom_normalize    
CTS-select： 
select_uchar_uchar
select_uchar_char
select_char_uchar
select_char_char
select_ushort_ushort
select_ushort_short
select_short_ushort
select_short_short
select_half_ushort
select_half_short
select_uint_uint
select_uint_int
select_int_uint
select_int_int
select_float_uint
select_float_int

Jules-Kong, 2024-10-10:
涉及到的用例还有 CTS-conversions；CTS-math_brute_force；

ziliangzl, 2024-10-12:
这个pocl的日志是在哪里看啊

Jules-Kong, 2024-10-12:
> 这个pocl的日志是在哪里看啊
设置 export POCL_DEBUG=all   
[image]

ziliangzl, 2024-10-24:
将pocl回滚到https://github.com/THU-DSP-LAB/pocl/commit/bb151c0a65a055b82a881b173ed9b742359869fb 能够通过测试。
也就是说问题出在后面的一个commit中：https://github.com/THU-DSP-LAB/pocl/commit/fbbf1d2f1e55a489c4195c4f4036894777424e07

closed 2024-10-24 by ziliangzl